anmaped/openfang #1: Newer buildroot & toolchains

jmtatsch, 2018-09-12:
Hi,

is there a reason why you are using such an old version of buildroot?
Have you seen the newer toolchains at https://github.com/tssva/ingenic-toolchains?

anmaped, 2018-09-12:
> 
> 
> Hi,
> 
> is there a reason why you are using such an old version of buildroot?
> Have you seen the newer toolchains at https://github.com/tssva/ingenic-toolchains?

@jmtatsch  I think so. First, kernel can only be compiled with gcc 4.7.x (maybe newer ports appear) and it is a good practice to use the same toolchain to compile all the distribution.
Second, the toolchains you point are based on glibc and not over ulibc.

jmtatsch, 2018-09-13:
I don't know.
Have you tried our fork https://github.com/Dafang-Hacks/v4l2rtspserver-master ?

anmaped, 2018-09-13:
> 
> 
> I don't know.
> Have you tried our fork https://github.com/Dafang-Hacks/v4l2rtspserver-master ?

Yes, but it is not stable at all. There are some pre-compiled libraries which may create some problems in my case (I have to recompile them) since the server shut down sporadically.

anmaped, 2018-09-20:
@jmtatsch Could you test the new rootfs? I didn't use yet the website and scripts but v4lrtspserver is now working.

jmtatsch, 2018-09-20:
Awesome, will do on the weekend

tssva, 2018-09-22:
> Second, the toolchains you point are based on glibc and not on ulibc as we expect.

The toolchains contain both glibc and uclibc.

anmaped, 2018-09-23:
@tssva Yes, this is a good point. We will see how to solve Ingenic MXU SIMD later.

Do you have some examples with OpenCV to test?

anmaped, 2018-09-24:
@jmtatsch Do you have the source of autonight?

jmtatsch, 2018-09-24:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Software/tree/master/autonight

There is also an open pullrequest for software autonight

anmaped, 2018-09-24:
> 
> 
> https://github.com/EliasKotlyar/Xiaomi-Dafang-Software/tree/master/autonight
> 
> There is also an open pullrequest for software autonight

Thanks. I will include it inside v4lrtspserver tools. Later we could make a proper package.

anmaped, 2018-09-25:
@jmtatsch The autonight tool is working in the current build: https://github.com/anmaped/openfang/tree/master/binaries/rootfs

jmtatsch, 2018-10-01:
Is the v4l2rtspserver completely unpatched?
Is it possible to include the stuff like  auto night, motor etc as submodule? that way it would be easier to track source changes & more transparent ... in future a build job could run the buildroot script when ever something has changed in the source repo

anmaped, 2018-10-01:
@jmtatsch Yes I have the same ideia. The source code that is available is the one already patched. I want to create the patches from the original source of v4l2rtspserver as you propose.

jmtatsch, 2018-10-21:
@anmaped I will do when i find some time, but I am quite busy with other things right now :(

jmtatsch, 2018-11-04:
@anmaped what are you plans regarding motion detection? That is arguably the most important feature of these cams :) You want to got the opencv route?

jmtatsch, 2018-11-04:
mqtt is working well by the way

anmaped, 2018-11-04:
Good. Could you provide a small tutorial how to configure it? We should
also describe the configuration parameters in the Gui.

anmaped, 2018-11-04:
> @anmaped what are you plans regarding motion detection? That is arguably the most important feature of these cams :) You want to got the opencv route?

Yes. I think opencv is the best option.

jmtatsch, 2018-11-04:
Basically I just added the configuration from dafanghacks mqtt.conf to /etc/mosquitto/mosquitto.conf
```
# Options for mosquitto_sub & mosquitto_pub                                    
USER=mqtt-user                                                                 
PASS=mqtt-password                                                             
HOST=my_mqtt_server_ip_adress                                                            
PORT=1883                                                                                                                                                                                                                                                     
```                                                                                                                                    
Then you can send a test message like that
mosquitto_pub -h my_mqtt_server_ip_adress  -m "test" -t "mylocation/test_topic"

Sure we can create a gui to change those parameters, however we would need to give cgi script root rights to write in /etc/. Root rights for anything web serving is almost never a good idea :)

As we are talking about root, can we add the sudo package to buildroot?

jmtatsch, 2018-11-04:
Concerning opencv motion detection, I would be thrilled if that would work out. 
However, i am afraid that resources are too limited for that. Ingenic wouldn't otherwise feel the need to provide highly optimised motion detection routines.

anmaped, 2018-11-05:
> As we are talking about root, can we add the sudo package to buildroot?

Fixed in 573788d0f31fd96255b197427aa242daa5c32194.

jmtatsch, 2018-11-21:
After the update /mnt/update somehow is still mounted, so the next update fails with:
`mount: mounting PPwPPw on /mnt/update failed: Device or resource busy`
although there is https://github.com/anmaped/openfang/blob/aeac4f0983762a0b36f3b46e339fe44b2da0ab15/fs/opt/autoupdate.sh#L20

jmtatsch, 2018-11-21:
Oh and I seem to have lost my wpa_supplicant.conf during the update :(

anmaped, 2018-11-22:
> 
> 
> Oh and I seem to have lost my wpa_supplicant.conf during the update :(

You can put the file in the second partition. It will be used automatically.

anmaped, 2018-11-22:
> 
> 
> After the update /mnt/update somehow is still mounted, so the next update fails with:
> `mount: mounting PPwPPw on /mnt/update failed: Device or resource busy`
> although there is
> 
> [openfang/fs/opt/autoupdate.sh](https://github.com/anmaped/openfang/blob/aeac4f0983762a0b36f3b46e339fe44b2da0ab15/fs/opt/autoupdate.sh#L20)
> 
> Line 20 in [aeac4f0](/anmaped/openfang/commit/aeac4f0983762a0b36f3b46e339fe44b2da0ab15)
>  umount /mnt/update/ 
> 
> using the same command manually works.
> Could this be a timing issue somehow?

Yes. The problem is that some tools are running, I should shutdown them first. This update script is outdated, let's see the new one in the next release, which you can find here https://github.com/anmaped/openfang/releases/download/rc05_01/autoupdate.sh.

anmaped, 2018-11-22:
@tssva Have you been able to compile it? Could you describe the dependencies in terms of the external environment?

closed 2019-04-06 by anmaped