anmaped/openfang #2: Current build uses soft floats

tssva, 2018-09-23:
The toolchain is built with `-msoft-float` as a default flag and buildroot is using `--with-float=soft` when compiling the rest of the tree. The xburst processors have a FPU and hard floats should be used to improve performance.

Also `-mno-fused-madd` should be used to avoid a bug in the FPU of xburst processors.

anmaped, 2018-09-23:
@tssva Thank you for your input. I was aware of that from the beginning since I have done it for purpose in order to make sure everything works well before enabling the hardware FPU.

Regarding the bug in the Xburst, do you have more information about it ?

tssva, 2018-09-23:
More info about the bug can be found at
https://groups.google.com/forum/m/#!topic/mips-creator-ci20/spDB2jjbizM.

anmaped, 2018-09-23:
Information from the group
> Is recommended to add -mno-fused-madd? Does this problem affect all the Ingenic SoCs?

> Looks like the answer is 'yes'.  However, on the default Debian distro, you will only need -mno-fused-madd if you also used the flag -march=mips32r2.  By default, the current distro only builds mips-2, which does not use madd.

> I do not know if this effects all Ingenic SoCs.  Earlier designs were mips32r1, so they are different.

> As a side note, in my testing so far, using the flag -march=mips32r2 gives a performance boost of about 13%.  I have not yet tested to see if this version of gcc supports the mxu SIMD instructions.

Apparently, for mips32r1 it is not required. But they said that mips32r2 give us 13% of boost on average.

anmaped, 2018-09-23:
Current ed1eeacca27e5999ec8a5d0d2ec0cfb22c86d944 commit solves the problem. FPU is enabled by default.

closed 2018-09-23 by anmaped

tssva, 2018-09-23:
The Ingenic toolchains have mips32r2 enabled by default.

Also more recent buildroot versions than currently being used have the
xburst architecture as an option. When enabled mips32r2, hard floats and
the no-fused-madd options are enabled.

anmaped, 2018-09-23:
Buildroot 2014.08 contains it. I will enable it later if needed. Until now I didn't see any advantage to go to mips32r2.