anmaped/openfang #6: Build fails on hosts with openssl 1.1.0

tssva, 2018-11-20:
If openssl 1.1.0 is installed on the build host compiling the Python _ssl and _hashlib modules fails causing the buildroot compile to fail.

The python build appears to be using the openssl headers installed on the build host instead of those of the buildroot openssl package. If the build host has openssl 1.1.0 installed this causes the build to fail because python 2.7 is not compatible with openssl 1.1.0.

anmaped, 2018-11-20:
I'm using `OpenSSL 1.1.0h-fips  27 Mar 2018`.

anmaped, 2018-11-20:
Is the problems building host-python-2.7.11 package ?

tssva, 2018-11-20:
My mistake. I didn't realize that it was building once for the host and then for the target platform. I assumed the error was happening when building for the target, but checking it is failing for host-python.

tssva, 2018-11-20:
host-openssl is not being built which is why host-python is using the openssl 1.1.0 headers in the host's /usr/include. Now to figure out why host-openssl isn't being built.

tssva, 2018-11-21:
Did some research and found the issue. host-python is built with ssl disabled; however, if the python build process detects the openssl header files during build it will try to build _ssl anyways in order to use the openssl hash routines for _hashlib. This fails when openssl 1.1.0 is installed. 

There are patches available which disable the check for openssl and has python use it's built-in version of the hash routines instead of openssl ones. I will submit a PR with the patches.

anmaped, 2018-11-21:
But do you know why in your system buildroot is not building host-openssl?

tssva, 2018-11-21:
host-python is built first because host-openssl isn't a dependency for it.

PR #8 (2018-11-21): Fix host-python build failing on systems with openssl 1.1.0 headers
Fixes #6
commit c8661af7 (2018-11-21): Fixes #6 (#8)
commit df5d98b7 (2018-11-21): Fixes #6
commit 4948293e (2018-11-21): Fix host-python build failing on systems with openssl 1.1.0 headers (Fixes #6)
closed 2018-11-21 by anmaped