anmaped/openfang #9: Install instructions need clarification

jkuepker, 2018-12-18:
A few questions on the install instructions:
1. How do I SSH to flash the u-boot on a stock Wyzecam T20?
2. Once I flash rootfs.ext3/2 what do I do?
3. How do I setup my Wifi network?
4. Can we get instructions for MacOS?

Also, in the windows instructions, the flasher is using ext2, but in the linux instructions, it is showing ext3.  Which one is it?

Thank you.

anmaped, 2018-12-19:
Regarding the first point you have to upload a modified bootloader using a fw hack or ask wyzecam for the ssh credential (xiaomi allows it but I'm not sure about wyzecam).

2) 3) You should do two partitions in the sdcard. One for the rootfs (ext3) and the second one in exfat to save the `wpa_supplicant.conf` file and other files of interest. It will connect to the wireless network according to the config intructions.

4) It is similar to the linux instructions. You have to use `fdisk` and `dd` utils.

Ext2 or ext3 image are the same. The journaling in the case of ext3 is enabled when starting the filesystem by the kernel.

akutruff, 2019-01-22:
I put wpa_supplicant.conf on the fat32 partition.  Are there any other files to put in that parttion?  Nothing is happening on the device.  Am I supposed to actually copy all the files on github into that partition just like Dafang-hacks, or is the ext2 partition simply ready as-is?

akutruff, 2019-01-22:
Also, I simply wrote the ext2 partition on Windows 10 using the utility you linked to.  I did notthing with ext3.  Is that correct?

anmaped, 2019-01-22:
@jkuepker Try to format the second partition as exfat instead of fat32.

akutruff, 2019-01-22:
I recreated the flash card from Ubuntu.  Still not working.  After untaring the files I noticed that rootfs.ext3 is a symbolic link to a non-existant ext2 file.  I flashed the rootfs.ext2 file to the first partition.

#This doesn't work
dd if=/path/to/image/rootfs.ext3 of=/dev/sdb1

Any ideas?  

I just saw that rc5 bootloader says untested... Does that mean it's never been tried by anyone?

anmaped, 2019-01-23:
> At the moment I'm just seeing a blinking orange light on the back of the camera, so I don't know if it's even getting to the wpa_supplicant stage. How do I know if things are going okay?

The led codes are:
- orange led blinking: not yet initialized
- orange led solid: successfully initialized
- blue led: microsd card activity

> Otherwise - Exfat isn't an option on windows 10... I tried NTFS just now as it showed in your screenshot.

I'm using windows 10 and I'm able format any sdcard to `exfat`.

> I recreated the flash card from Ubuntu. Still not working. After untaring the files I noticed that rootfs.ext3 is a symbolic link to a non-existant ext2 file. I flashed the rootfs.ext2 file to the first partition.

ext2 file is the correct file to flash. As already said, `ext2` or `ext3` images are the same (one inks to the other). The journaling in the case of `ext3` is enabled when starting the filesystem by the kernel.

>  dd if=/path/to/image/rootfs.ext3 of=/dev/sdb1
You should modify `/dev/sdb1` to the correct device and use the rootfs.ext2 image instead of rootfs.ext3.

> I just saw that rc5 bootloader says untested... Does that mean it's never been tried by anyone?

It's a safety warning.

akutruff, 2019-01-23:
Thanks for the replies.  

What do you mean modify /dev/sdb1 to the correct device?  That’s the first partition on the SD card.

anmaped, 2019-01-23:
@akutruff I mean that you could have other devide such as sdc1 or sdd1 etc...

akutruff, 2019-01-23:
Oh, heh.  Yeah, sorry, I misunderstood.  I verified sdb was by usb drive and partitioned it to a 5GB type 7 (sdb1) and a 32GB type 7 (sdb2.)

Well, I'm kind of stuck now.  Not sure what I can try next.

anmaped, 2019-01-23:
@akutruff Which camera are you using? Have you flashed the bootloader?

akutruff, 2019-01-23:
Wyze V2.  Yep.  I flashed the bootloader.  I gained access to the device by using the dafang-hacks project.  I then followed your instructions.

akutruff, 2019-01-23:
Are there more instructions that are assumed as obvious that I'm possibly missing?  For example, you mentioned wpa_supplicant.conf in this thread and that is not mentioned in the installation documentation.  I have a single file on partition 2 - the wpa_supplicant.conf.

anmaped, 2019-01-28:
@akutruff Do you have a chance to use a serial interface over the camera?

akutruff, 2019-01-28:
I'd have to solder, correct?  Reluctant to do that until I know the full instructions to install the firmware and where to put wpa_supplicant.conf.

I'm happy to continue working through issues on this project.  However, I need clear documentation of the setup process.  Please document everything, starting completely fresh from a factory camera.  Write down absolutely every step and command you type to install everything from start to finish.  Assume the person reading your instructions knows practically nothing.  That's the only way for an installation like this to possibly succeed for a new user.  

So many open source projects do not document setup instructions for new users to the level of detail needed.  Thanks in advance.

anmaped, 2019-02-07:
> I'd have to solder, correct? Reluctant to do that until I know the full instructions to install the firmware and where to put wpa_supplicant.conf.

Unfortunately, yes. We can also write stuff to the sdcard in order to debug the problem.

> So many open source projects do not document setup instructions for new users to the level of detail needed. Thanks in advance.

You are right, but sometimes we also need help from the community. You can help us to give such a contribution. Could you elaborate on the steps you have done ?

akutruff, 2019-02-07:
I followed all the steps in the instructions and then had to fill in the blanks above.  

I recommend you start completely from from a new camera and try to follow the instructions precisely as written through the eyes of a new user.  If you type something that's not in the instructions, I recommend putting them in there.  I'm guessing you'll find that there's steps that were assumed.

akutruff, 2019-02-07:
Also, considering I bricked my camera, I think I'm doing my part as a member of the community.  ; )

anmaped, 2019-02-07:
> 
> 
> Also, considering I bricked my camera, I think I'm doing my part as a member of the community. ; )

What do you mean by brick?

anmaped, 2019-02-07:
Files [doc/install.md](../tree/master/doc/install.md) and [doc/troubleshooting.md](../tree/master/doc/troubleshooting.md) have been updated.

closed 2019-02-09 by anmaped