anmaped/openfang #13: v4l2rtspserver not working

llimz, 2019-02-02:
Hello,

My camera is now almost working. The model is Mijia 1080P camera (with the white back). I've got the network connection ok, SSH connection working too. But the video streaming is not working. After the boot process, I'm connecting through SSH and I don't see any v4l2rtspserver process running (via the command line 'ps aux' for example). 

I then try to launch the process manually (with 'su' first just in case). If I launch `v4l2rtspserver-v0.0.8 -W 1280 -H 960', I got an error [1]+  Illegal instruction. See the log below:

`$ v4l2rtspserver-v0.0.8 -W 1280 -H 960 &
$ 2019-02-02 20:06:43.048 (   0.001s) [main thread     ]               main.cpp:450      0| === Dumping environment variables ===
2019-02-02 20:06:43.051 (   0.004s) [main thread     ]               main.cpp:453      0| USER=admin
2019-02-02 20:06:43.053 (   0.006s) [main thread     ]               main.cpp:453      0| SSH_CLIENT=192.168.0.50 53968 22
2019-02-02 20:06:43.055 (   0.008s) [main thread     ]               main.cpp:453      0| SHLVL=2
2019-02-02 20:06:43.056 (   0.010s) [main thread     ]               main.cpp:453      0| OLDPWD=/home/admin
2019-02-02 20:06:43.058 (   0.011s) [main thread     ]               main.cpp:453      0| HOME=/root
2019-02-02 20:06:43.059 (   0.012s) [main thread     ]               main.cpp:453      0| SSH_TTY=/dev/pts/0
2019-02-02 20:06:43.060 (   0.013s) [main thread     ]               main.cpp:453      0| PAGER=/bin/more 
2019-02-02 20:06:43.060 (   0.014s) [main thread     ]               main.cpp:453      0| PS1=$ 
2019-02-02 20:06:43.061 (   0.014s) [main thread     ]               main.cpp:453      0| LOGNAME=admin
2019-02-02 20:06:43.062 (   0.015s) [main thread     ]               main.cpp:453      0| TERM=xterm-256color
2019-02-02 20:06:43.063 (   0.016s) [main thread     ]               main.cpp:453      0| PATH=/bin:/sbin:/usr/bin:/usr/sbin
2019-02-02 20:06:43.064 (   0.017s) [main thread     ]               main.cpp:453      0| SSH_AUTH_SOCK=/tmp/dropbear-623cb2e7/auth-d146cd80-8
2019-02-02 20:06:43.064 (   0.017s) [main thread     ]               main.cpp:453      0| SHELL=/bin/sh
2019-02-02 20:06:43.065 (   0.019s) [main thread     ]               main.cpp:453      0| PWD=/etc/init.d
2019-02-02 20:06:43.066 (   0.019s) [main thread     ]               main.cpp:453      0| SSH_CONNECTION=192.168.0.50 53968 192.168.0.69 22
2019-02-02 20:06:43.067 (   0.020s) [main thread     ]               main.cpp:453      0| EDITOR=/bin/vi
2019-02-02 20:06:43.068 (   0.021s) [main thread     ]               main.cpp:456      0| === end ===
2019-02-02 20:06:43.081 (   0.034s) [main thread     ]         ImpEncoder.cpp:685      0| Try to read extra configuration on /usr/bin/v4l2rtspserver-v0.0.8.ini
2019-02-02 20:06:43.083 (   0.036s) [main thread     ]         ImpEncoder.cpp:688      0| Can't load 'v4l2rstpserver.ini'. Enable default paths
2019-02-02 20:06:43.314 (   0.267s) [main thread     ]         ImpEncoder.cpp:993      0| Found Sensor with ID:1
2019-02-02 20:06:43.318 (   0.271s) [main thread     ]         ImpEncoder.cpp:1007     0| Found Sensor with Name:ps5250
2019-02-02 20:06:43.859 (   0.812s) [main thread     ]         ImpEncoder.cpp:1241     0| encoderMode: 3
2019-02-02 20:06:43.871 (   0.824s) [main thread     ]         ImpEncoder.cpp:1280     0| Using VBR mode.
2019-02-02 20:06:43.873 (   0.826s) [main thread     ]         ImpEncoder.cpp:1285     0| Set initial bitrate (from sharedmem) to 2666
i264e[info]: profile Constrained Baseline, level 3.2
2019-02-02 20:06:43.925 (   0.878s) [update_thread   ]                OSD.cpp:22       0| [OSD]: Created OSD(1248, 0, 32, 32, 0)
2019-02-02 20:06:43.925 (   0.878s) [update_thread   ]                OSD.cpp:22       0| [OSD]: Created OSD(0, 0, 1280, 10, 1)
2019-02-02 20:06:43.927 (   0.880s) [update_thread   ]         ImpEncoder.cpp:181      0| Attempt to changed Bitrate to 2666
2019-02-02 20:06:43.931 (   0.884s) [update_thread   ]         ImpEncoder.cpp:199      0| Setting VBR maxBitrate.
2019-02-02 20:06:43.933 (   0.887s) [update_thread   ]         ImpEncoder.cpp:216      0| Attempt to changed fps to 25,1
2019-02-02 20:06:43.934 (   0.888s) [update_thread   ]         ImpEncoder.cpp:250      0| Font name:/usr/share/fonts/NotoSans-Regular.ttf
2019-02-02 20:06:43.955 (   0.908s) [update_thread   ]         ImpEncoder.cpp:262      0| Changed OSD font
2019-02-02 20:06:43.973 (   0.926s) [update_thread   ]         ImpEncoder.cpp:292      0| Max font bitmap height is 31 and baseline offset is -6
2019-02-02 20:06:43.976 (   0.929s) [update_thread   ]         ImpEncoder.cpp:294      0| Changed OSD size
2019-02-02 20:06:43.981 (   0.934s) [update_thread   ]         ImpEncoder.cpp:314      0| Deactivate motion
2019-02-02 20:06:43.983 (   0.936s) [update_thread   ]         ImpEncoder.cpp:338      0| Display motion OSD color=-1
2019-02-02 20:06:44.073 (   1.026s) [update_thread   ]         ImpEncoder.cpp:379      0| Failed to read the JPEG stream: Polling stream timeout
2019-02-02 20:06:44.415 (   1.368s) [DeviceSource    ]       DeviceSource.cpp:91       0| begin thread
2019-02-02 20:06:44.421 (   1.374s) [main thread     ]               main.cpp:566      0| Create ALSA Source.../dev/dsp
2019-02-02 20:06:44.422 (   1.375s) [main thread     ]        ALSACapture.cpp:96       0| Audio In GetPubAttr samplerate : 44100
2019-02-02 20:06:44.422 (   1.375s) [main thread     ]        ALSACapture.cpp:97       0| Audio In GetPubAttr   bitwidth : 16
2019-02-02 20:06:44.422 (   1.375s) [main thread     ]        ALSACapture.cpp:98       0| Audio In GetPubAttr  soundmode : 1
2019-02-02 20:06:44.422 (   1.375s) [main thread     ]        ALSACapture.cpp:99       0| Audio In GetPubAttr     frmNum : 50
2019-02-02 20:06:44.422 (   1.375s) [main thread     ]        ALSACapture.cpp:100      0| Audio In GetPubAttr  numPerFrm : 8820
2019-02-02 20:06:44.423 (   1.376s) [main thread     ]        ALSACapture.cpp:101      0| Audio In GetPubAttr     chnCnt : 1

[1]+  Illegal instruction        v4l2rtspserver-v0.0.8 -W 1280 -H 960`

If I try to with other parameters, I've got only the sound, no picture. I got several errors (maybe one per second) "Failed to read the Jpeg Stream". See below to get the log:

` $ v4l2rtspserver-v0.0.8 -W 1280 -H 960 &
$ 2019-02-02 20:06:43.048 (   0.001s) [main thread     ]               main.cpp:450      0| === Dumping environment variables ===
2019-02-02 20:06:43.051 (   0.004s) [main thread     ]               main.cpp:453      0| USER=admin
2019-02-02 20:06:43.053 (   0.006s) [main thread     ]               main.cpp:453      0| SSH_CLIENT=192.168.0.50 53968 22
2019-02-02 20:06:43.055 (   0.008s) [main thread     ]               main.cpp:453      0| SHLVL=2
2019-02-02 20:06:43.056 (   0.010s) [main thread     ]               main.cpp:453      0| OLDPWD=/home/admin
2019-02-02 20:06:43.058 (   0.011s) [main thread     ]               main.cpp:453      0| HOME=/root
2019-02-02 20:06:43.059 (   0.012s) [main thread     ]               main.cpp:453      0| SSH_TTY=/dev/pts/0
2019-02-02 20:06:43.060 (   0.013s) [main thread     ]               main.cpp:453      0| PAGER=/bin/more 
2019-02-02 20:06:43.060 (   0.014s) [main thread     ]               main.cpp:453      0| PS1=$ 
2019-02-02 20:06:43.061 (   0.014s) [main thread     ]               main.cpp:453      0| LOGNAME=admin
2019-02-02 20:06:43.062 (   0.015s) [main thread     ]               main.cpp:453      0| TERM=xterm-256color
[... 80 lines cut ...]
2019-02-02 20:09:31.751 (   1.383s) [main thread     ]        ALSACapture.cpp:98       0| Audio In GetPubAttr  soundmode : 1
2019-02-02 20:09:31.751 (   1.383s) [main thread     ]        ALSACapture.cpp:99       0| Audio In GetPubAttr     frmNum : 50
2019-02-02 20:09:31.751 (   1.383s) [main thread     ]        ALSACapture.cpp:100      0| Audio In GetPubAttr  numPerFrm : 8820
2019-02-02 20:09:31.751 (   1.383s) [main thread     ]        ALSACapture.cpp:101      0| Audio In GetPubAttr     chnCnt : 1
2019-02-02 20:09:31.772 (   1.404s) [DeviceSource    ]       DeviceSource.cpp:91       0| begin thread
2019-02-02 20:09:32.188 (   1.820s) [main thread     ]        ALSACapture.cpp:128      0| Audio In GetChnParam usrFrmDepth : 50
2019-02-02 20:09:32.202 (   1.834s) [main thread     ]        ALSACapture.cpp:155      0| Audio In GetVol    vol : 70
2019-02-02 20:09:32.207 (   1.838s) [main thread     ]        ALSACapture.cpp:390      0| Disabled highfilter
[... cut ...]
2019-02-02 20:09:37.011 (   6.642s) [update_thread   ]         ImpEncoder.cpp:379      0| Failed to read the JPEG stream: Polling stream timeout
2019-02-02 20:09:39.011 (   8.642s) [update_thread   ]         ImpEncoder.cpp:379      0| Failed to read the JPEG stream: Polling stream timeout
^C2019-02-02 20:09:39.417 (   9.049s) [main thread     ]               main.cpp:59     ERR| SIGINT
2019-02-02 20:09:39.418 (   9.050s) [main thread     ]               main.cpp:670      0| Exiting....
`

Any idea? Thank you.

anmaped, 2019-02-04:
Could you try 73bd1cfa9e3dadc1a67529595d8b6c8543cb2d5b. It fixes your problem!

anmaped, 2019-02-04:
Just clean v4l2rtspserver using
```
cd _build/buildroot-2016.02/
make v4l2rtspserver-dirclean # to clean the old compilation of v4l2rtspserver
cd ../../
./buildopenfang.sh # to compile it into the image again
```

anmaped, 2019-02-04:
Don't forget to generate the init file `/usr/bin/v4l2rtspserver-v0.1.3.ini` as follows:
```
# default configuration for v4l2rtspserver
[Hardware]
SensorName=ps5250
SensorAddr=0x48

[Video]
Width=1280
Height=960
Fps=25
Bitrate=4000

[RTSP]
Port=8554
Url=unicast
```

llimz, 2019-02-04:
I got the errors below when trying to compile. It seems that there is still old references to the previous v4l2rtspserver 

Applying sensor_fix.patch using patch: 
can't find file to patch at input line 4
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff -Nura v4l2rtspserver-v0.0.8/src/ImpEncoder.cpp v4l2rtspserver-v0.0.8_mod/src/ImpEncoder.cpp
|--- v4l2rtspserver-v0.0.8/src/ImpEncoder.cpp	2018-09-20 00:40:42.000000000 +0200
|+++ v4l2rtspserver-v0.0.8_mod/src/ImpEncoder.cpp	2019-01-29 23:30:36.227767100 +0100
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix sensor_fix.patch!
package/pkg-generic.mk:159 : la recette pour la cible « /media/PhoneGap/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/.stamp_patched » a échouée
make: *** [/media/PhoneGap/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/.stamp_patched] Erreur 1

anmaped, 2019-02-04:
Good news! Thanks for the effort :) May I close the issue ?

closed 2019-02-04 by llimz