anmaped/openfang #14: IP address changing every reboot

llimz, 2019-02-04:
Hello,

Just in case it helps. With my mijia "white back" camera, the IP address changes on every reboot. After some investigations it seems that the partition read to find the MAC address is empty with this camera. 

If I run:

```
umount /params
mount -t jffs2 /dev/mtdblock9 /params
```

The params folder is empty.

anmaped, 2019-02-04:
Yes. We have to read the EFuse bits of the wireless chip in order to get the right MAC address.

anmaped, 2019-02-22:
Commmit dd03c77a03ed773ede2e11c554b3e0a0402edfab Fixes #14.

closed 2019-02-22 by anmaped

llimz, 2019-03-13:
Hello, I just tested it with last version commited since the 12th of march and it is working fine. Nice job by the way ! The new way of setting up the camera settings including Wifi SSID and password is very user friendly (via the web interface).

I still cannot get pictures/video from the camera with v4l2rtspserver but I assume it is normal until a new version/fix is used? I have only 1 frame on the web interface, it is not refreshed. Is there any way to get if possible an mjpeg stream or something?

anmaped, 2019-03-14:
@llimz Everything is working now even the mjpeg stream. Ask in the openfang gitter chat if you have some issues.