anmaped/openfang #15: v4l2rtspserver v0.1.3 HLS

llimz, 2019-02-04:
Hello,

With this version of v4l2rtspserver, does the HLS stream working? I cannot find the correct parameter to put in the .ini file (and I don't know if it exists or not). In the previous version, I read that there was a parameter "-S" in the command line to enable the HLS stream. Thank you.

anmaped, 2019-02-05:
Yes, it's that.

puddly, 2019-03-07:
@anmaped I've had the pleasure of working somewhat extensively with `libimp` and I'd love to finally delete it.

The most interesting part of `libimp` is Ingenic's H264 encoder, which appears to be just a stripped-down version of [x264](https://www.videolan.org/developers/x264.html) (I hope properly licensed) with `x264_` replaced with `i264e_` in most of the function names and patches to make it utilize the accelerated H264 encoder.

For comparison, I get about 9fps at 640x360 (with the `ultrafast` + `zerolatency` presets) when piping uncompressed frames captured by `libimp` into an unoptimized `libx264` (which doesn't have any speedups for MIPS at all). 1920x1080 encodes at 1.4fps. The upside is that you can actually tune the encoder to produce NAL units properly sized for RTP streaming, so UDP streaming will finally be reliable.

The good news is that the source code for the patches to x264 for a similar, older platform [are available](https://github.com/ljjielei/m200_app/tree/master/x264/encoder) so in theory there's enough code floating around to talk to the hardware encoder and get rid of `libimp` entirely.

anmaped, 2019-03-20:
@puddly Good that you also have the same insight as me! I have started with some experiments with `gdb` and `strace`. As far as I understood there is a subdevice that the kernel manages where all the configurations are done and we have to know how to perform these configurations. I think debugging it extensively will help us.

Could you describe your setup? Have you tried to use gcc 5 to compile libx264 with SIMD instruction set?

> The upside is that you can actually tune the encoder to produce NAL units properly sized for RTP streaming, so UDP streaming will finally be reliable.

Have you tested it ?

> The good news is that the source code for the patches to x264 for a similar, older platform are available so in theory there's enough code floating around to talk to the hardware encoder and get rid of libimp entirely.

I guess that the hardware encoder is old and shared by other SOCs.

puddly, 2019-03-22:
> Could you describe your setup?

I wrote a small RTSP server that interfaces with libimp and libx264 to debug the RTSP problems people seem to be having. I'll clean up the build environment and open source it when I have time.

> Have you tried to use gcc 5 to compile libx264 with SIMD instruction set?

Thanks, I wasn't aware the new toolchain supported MSA. I am able to compile it but my binary won't link to both libimp and libx264 (something about `-mmsa` enabling floating point stuff that's incompatible with libimp). Once I figure out a way to fix that I'll let you know if it speeds anything up significantly.

> Have you tested it ?

I have, but 640x360@9fps and 1920x1080@1fps aren't very realistic settings so I can't be 100% sure.

esarigul, 2019-05-01:
@anmaped I am a complete noob here I downloaded your github compiled it for different camera runs on T20 SoC (not Dafang or other models on its web server (setmodel script needed to be modified), it required some tweaking ( I found all corresponding gpio and trying to modify init scripts accordingly (I am still working on it) but it would be nice to have in web interface to other other T20 SoC based devices and have config file (you used nvram method but It does not work in my camera, I am trying to modify uboot-env partition and use fw_nvram script to use get part (set part already do nothing), but my problem is v4l2rtspserver, when I compile it it defaults to v0.0.8 but I want to compile v0.1.3 version how can I modify buildroot to compile that version? v4l2rtspserver-v0.0.8 does not work it tries to establish H264 stream with VLC mobile app, it disconnects after that? thank you in advance

anmaped, 2019-05-01:
@esarigul please open a new issue. Your problem doesn't relate to this one. Before opening it please explain the hardware and the things you want to achieve./issues found

esarigul, 2019-05-01:
will do thanks

raulvasquez, 2019-10-25:
Can v4l2rtspserver be upgraded inline with the .19 release? I found this repo from dafang-hacks, but was not sure if it would be compatible with buildroot - 
https://github.com/Dafang-Hacks/Main/tree/master/v4l2rtspserver-master