anmaped/openfang #16: Flash dump - with or without bootloader?

therosss, 2019-02-07:
Does the Flash dump you supplied @ openfang/doc/sxj02ZM/ dump.tar.xz already havee a bootloader onboard? I'm missing the info if it is untouched or modded.

anmaped, 2019-02-07:
Hi @therosss! As the name suggests is just the original dump, otherwise it should be a custom FW.

therosss, 2019-02-07:
Ooooh. That is the reason why my cam did not boot into the custom Firmware.
Thank you very much for the Response.
Could you link me to a compiled bootloader which I can burn on my SOP?
Feel free to close this issue as it doesn't seem to be one.

anmaped, 2019-02-07:
Here it is [u-boot-lzo-with-spl_t20_64M.zip](https://github.com/anmaped/openfang/files/2842929/u-boot-lzo-with-spl_t20_64M.zip).

therosss, 2019-02-08:
Great! Thank you very much. I'm slightly confused as the SOP is actually a 128mbit one.

After erasing the SOP and flashing it only with the u-boot-lzo-with-spl_t20_64M.bin file, the cam is dead again. I wonder what I am doing wrong here :D

anmaped, 2019-02-08:
I guess you are not using the right rootfs (which one are you using?) since the bootloader doesn't change the leds. If you can check the serial you'll see that it is working.

therosss, 2019-02-08:
Since I did not yet compile much, I used the rootfs.ext2 from "release candidate 5 - openfang 01" which you compiled.
I'll give it a try with compiling the current repo and writing the rootfs to the SD Card.

anmaped, 2019-02-08:
You can use this file [rootfs.ext2_04022019.xz](https://ufile.io/jtv42).
I will do the new rc6 release soon!

therosss, 2019-02-08:
Thank you @anmaped After all it was the rootfs that was not right.

The cam booted now and It's connecting to the Wifi etc.

Still having a lot of issues like WebUI not working properly, LEDs not functioning, poor image quality and IR-LEDs not working either, but that'S something which I wasn't aiming at first.
Do you want me to open an issue for each case, or shall I first wait a bit?

anmaped, 2019-02-08:
Just wait a bit. Some issues are already solved.

anmaped, 2019-02-08:
@therosss What do you mean by "LEDs not functioning"? Are blue and yellow LEDs not blinking etc?

therosss, 2019-02-08:
The Orange LED is constantly on.
And the controls of the web GUI don't work

anmaped, 2019-02-22:
Fix dd03c77a03ed773ede2e11c554b3e0a0402edfab.

closed 2019-02-22 by anmaped