anmaped/openfang #17: Openfang vs Dafang-hacks

JelmerT, 2019-02-07:
[Dafang-Hacks](https://github.com/Dafang-Hacks/rootfs) seems like a very similar effort to [Openfang](https://github.com/anmaped/openfang/)

Which one should I choose to flash and contribute to?
Wouldn't it be beneficial to both to combine forces and focus on one of both?
@anmaped and I think @EliasKotlyar among others

anmaped, 2019-02-07:
Hi @JelmerT !

Good that you ask that. We are not disjoint forces since we are sharing information between both projects, but I will try to clarify some points. For some cases you just can choose Dafang-Hacks but if you are interested in OpenCV and on a platform that you want to modify then you should choose Openfang.

As one of the Dafang-hacks contributors said in October last year

> If you have flashed the custom uboot from Dafang-Hacks, you can also try the new rootfs (my pull request) or openfang rootfs directly. No need to flash the rom...
https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/issues/573#issuecomment-427738793

Or yet:
> @arkhub you may want to try my pull request [Dafang-Hacks/rootfs#3](https://github.com/Dafang-Hacks/rootfs/pull/3) most stuff should work. Or you can have a look at https://github.com/anmaped/openfang promising but not yet fully functional.
https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/issues/575#issuecomment-435716931

This means that both are interchangeble somehow. As a reminder, just now openfang has support for Xiaomi Mijia 2018 and Dafang-Hacks doesn't support it yet.

Openfang is a bit different from Dafang-hacks. Dafang Hacks as the name says is a hack and Openfang is built on top of Buildroot (it is a kernel, a bootloader, and a toolchain), which allows you to install and modify many other packages that you are not able to do with Dafang-Hacks (like using ubuntu or fedora on embedded systems), and Openfang is a complete opensource suite.

Buildroot is much wider and general tool for extending and developing new features and you could include what you want by just changing the configuration files, create a new firmware, adjust many other parameters in a more efficient and easy way.

In terms of packages, Openfang just uses the modified v4l2rtspserver provided from Dafang-Hacks but some changes are planned to be applied soon.

To sum-up Openfang is not a fork of Dafang-Hacks is more like a normal embeeded and extensible paltform for devices (e.g., cameras) using Ingenic SOCs that just born because of the limitations of a hack oriented development project like Dafang-Hacks.

JelmerT, 2019-02-07:
Thanks for taking the time to explain this @anmaped!
I conclude that the openfang is more flexible towards future/other devices and (currently) the most active, although being based on some original efforts made with the Dafang-hacks. 👍

closed 2019-02-07 by JelmerT