anmaped/openfang #21: Build files

Anirudhdh, 2019-03-15:
I run the buildopenfang.sh script on Debian GCC 6.3.0 and get the following errors:
`[ 56%] Building CXX object CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/MulticastServerMediaSubsession.cpp.o
[ 59%] Building CXX object CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/ServerMediaSubsession.cpp.o
[ 62%] Building CXX object CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/ImpCapture.cpp.o
[ 65%] Building CXX object CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/main.cpp.o
[ 68%] Building CXX object CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/MJPEGVideoSource.cpp.o
In file included from /home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/src/MJPEGVideoSource.cpp:14:0:
/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/inc/MJPEGVideoSource.h:15:30: fatal error: JPEGVideoSource.hh: No such file or directory
compilation terminated.
In file included from /home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/src/ServerMediaSubsession.cpp:14:0:
/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/inc/ServerMediaSubsession.h:20:24: fatal error: liveMedia.hh: No such file or directory
compilation terminated.
In file included from /home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/inc/ImpCapture.h:4:0,
                 from /home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/src/ImpCapture.cpp:1:
/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/inc/DeviceSource.h:22:24: fatal error: liveMedia.hh: No such file or directory
compilation terminated.
In file included from /home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/inc/MulticastServerMediaSubsession.h:12:0,
                 from /home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/src/MulticastServerMediaSubsession.cpp:10:
/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/inc/ServerMediaSubsession.h:20:24: fatal error: liveMedia.hh: No such file or directory
compilation terminated.
/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/src/main.cpp:29:36: fatal error: BasicUsageEnvironment.hh: No such file or directory
compilation terminated.
CMakeFiles/v4l2rtspserver-v0.1.3.dir/build.make:62: recipe for target 'CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/MulticastServerMediaSubsession.cpp.o' failed
make[3]: *** [CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/MulticastServerMediaSubsession.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
CMakeFiles/v4l2rtspserver-v0.1.3.dir/build.make:182: recipe for target 'CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/MJPEGVideoSource.cpp.o' failed
make[3]: *** [CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/MJPEGVideoSource.cpp.o] Error 1
CMakeFiles/v4l2rtspserver-v0.1.3.dir/build.make:110: recipe for target 'CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/ServerMediaSubsession.cpp.o' failed
make[3]: *** [CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/ServerMediaSubsession.cpp.o] Error 1
CMakeFiles/v4l2rtspserver-v0.1.3.dir/build.make:134: recipe for target 'CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/main.cpp.o' failed
make[3]: *** [CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/main.cpp.o] Error 1
CMakeFiles/v4l2rtspserver-v0.1.3.dir/build.make:206: recipe for target 'CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/ImpCapture.cpp.o' failed
make[3]: *** [CMakeFiles/v4l2rtspserver-v0.1.3.dir/src/ImpCapture.cpp.o] Error 1
make[3]: Leaving directory '/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3'
CMakeFiles/Makefile2:291: recipe for target 'CMakeFiles/v4l2rtspserver-v0.1.3.dir/all' failed
make[2]: *** [CMakeFiles/v4l2rtspserver-v0.1.3.dir/all] Error 2
make[2]: Leaving directory '/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3'
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3'
package/pkg-generic.mk:195: recipe for target '/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/.stamp_built' failed
make: *** [/home/tt/openfang/_build/buildroot-2016.02/output/build/v4l2rtspserver-v0.1.3/.stamp_built] Error 2`
How to solve this issuse?

Cabalist, 2019-03-15:
Could you use \`\`\` to enclose the error message?  The formatting is all messed up without newlines. :)

Anirudhdh, 2019-03-15:
> Could you use ``` to enclose the error message? The formatting is all messed up without newlines. :)

Fixit, sanx.

Cabalist, 2019-03-15:
It looks like `live555` isn't being downloaded or put in the right place by the build script.

Silly question but are you building this on a machine with internet access?

Anirudhdh, 2019-03-16:
> It looks like `live555` isn't being downloaded or put in the right place by the build script.
> 
> Silly question but are you building this on a machine with internet access?
```find / -iname liveMedia.hh
/usr/include/liveMedia/liveMedia.hh
/home/tt/openfang/_build/buildroot-2016.02/output/host/usr/mipsel-ingenic-linux-uclibc/sysroot/usr/include/liveMedia/liveMedia.hh
/home/tt/openfang/_build/buildroot-2016.02/output/build/live555-2016.01.29/liveMedia/include/liveMedia.hh
/home/tt/openfang/_build/buildroot-2016.02/output/target/usr/include/liveMedia/liveMedia.hh```
Internet is available, and liveMedia.hh looks downloaded, but buildopenfang.sh can't find it

anmaped, 2019-03-16:
How do you managed to enable v4l2rtspserver version 0.1.3 ?

Anirudhdh, 2019-03-17:
> How do you managed to enable v4l2rtspserver version 0.1.3 ?
> 
> Have you tried the docker container?

Im just installed liblivemedia-dev from debian repo. Probably v4l2rtspserver is from there. Dont use docker

anmaped, 2019-03-18:
Could I close the issue then?

Anirudhdh, 2019-03-18:
> Could I close the issue then?

Issue that I report is still not solved. Probably, the decision is obvious, but I am a noob in Linux. Could you indicate a exact solution?

anmaped, 2019-03-19:
Try to use docker then.

Anirudhdh, 2019-03-19:
> Try to use docker then.
```
root@qkg:/home/tt# docker build -f ./Dockerfile -t openfang/container .
Sending build context to Docker daemon   7.11GB
Error response from daemon: unexpected error reading Dockerfile: read /var/lib/docker/tmp/docker-builder060510054/Dockerfile: is a directory
```
Errors what i get.

Anirudhdh, 2019-03-19:
> Try to use docker then.

```
tt@qkg:~$ docker build -f ./Dockerfile -t openfang/container .
unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/tt/Dockerfile: no such file or directory
tt@qkg:~$ mkdir /home/tt/Dockerfile

tt@qkg:~$ docker build -f ./Dockerfile -t openfang/container .
Sending build context to Docker daemon   7.11GB
Error response from daemon: unexpected error reading Dockerfile: read /var/lib/docker/tmp/docker-builder922711618/Dockerfile: is a directory
```
I get follow errors with docker installation.

linkiln, 2019-03-25:
There is a problem in 0.1.3 with source of live555 package. You need to decompress and manually change location. After that create archive of v4l2rtspserver-v0.1.3. Also there is problem with 3 libraries libsharedmem.so, libwrap.so and libfilter.so, you need to put them in /usr/lib.

anmaped, 2019-03-25:
v4l2rtspserver have done some changes in the repository (live555), you can check it there. However, 0.1.3 it's just not stable as 0.0.8 so I don't recommend to use it for now.