anmaped/openfang #27: Purple tint in video and images

AbcHeng, 2019-04-01:
Hi,

On my Wyze Pan Camera, the video and images have a purple tint when using the openfang repo.

I don't have this issue when using the dafang-hacks repo however.

Is there a configuration setting to fix this?

anmaped, 2019-04-01:
IR switch or wait by night and it will set it up well.

AbcHeng, 2019-04-01:
Hmm, it shows that they're off in the web UI but I'll check again.

anmaped, 2019-04-01:
This is a problem with reverse pinouts and some cameras have inverted gpios, others do not. I will fix this soon. Try to turn on and off again.

anmaped, 2019-04-02:
Is it solved ?

AbcHeng, 2019-04-03:
Hi,

Yes thanks,, but I had to do a few things.

1) For auto night detection, the auto-night-detection file wasn't under the controlscripts folder.

2) IR cut on is what was able to get it to stop the purple hue / tint.

3) There's quite a few other questions and problems I have. Is there a better way to communicate, or should I just post here on Github.

anmaped, 2019-04-06:
@abcheng Why you need auto-night-detection script... auto night daemon is running.

Try to use our gitter channel.

abcheng, 2019-04-06:
@anmaped, you mentioned auto by night, which I thought was auto night detection. The only thing I saw was a toggle button which calls the auto night detection script so I copied that over from dafang hacks. 

Sounds good, I'll try the gitter channel.

anmaped, 2019-05-05:
Fixed b869345a5f43f249c6bcf50cc2dc8e8665215ac1.

closed 2019-05-05 by anmaped