anmaped/openfang #33: Compiling v4l2rtspserver-v0.1.3 instead of default version which seems buggy

esarigul, 2019-05-01:
@anmaped I am a complete noob here I downloaded your github compiled it for different camera runs on T20 SoC (not Dafang or other models on its web server (setmodel script needed to be modified), it required some tweaking ( I found all corresponding gpio and trying to modify init scripts accordingly (I am still working on it) but it would be nice to have in web interface to other other T20 SoC based devices and have config file (you used nvram method but It does not work in my camera, I am trying to modify uboot-env partition and use fw_nvram script to use get part (set part already do nothing), but my problem is v4l2rtspserver, when I compile it it defaults to v0.0.8 but I want to compile v0.1.3 version how can I modify buildroot to compile that version? v4l2rtspserver-v0.0.8 does not work it tries to establish H264 stream with VLC mobile app, it disconnects after that? thank you in advance

esarigul, 2019-05-01:
Forgot to add hardware info:
SoC: T20L (64MB DDR ram)
SPI flash is W25Q80 (8 MB)
no PTZ control

anmaped, 2019-05-01:
So you should do a pull request with the changes, and explain which model are you using. Again v4l2rtspserver-v0.0.8 is the default version and v4l2rtspserver-v0.1.3 is not stable. So use 0.0.8.

esarigul, 2019-05-01:
@anmaped Can you tell me to steps compile v0.1.3 since v0.0.8 does not work for me, it disconnects at the start.
thx fpr your time

anmaped, 2019-05-01:
Then, 0.1.3 will not work as well.

esarigul, 2019-05-01:
I check info in /var/log/messages it does not write anything there but in debug console puts the following info, if there is any other log info that I am not aware of,
Debug console output is when I try to make connection
[main thread     ]ServerMediaSubsession.c:25       0| Creating Source...                                              
[main thread     ]ServerMediaSubsession.c:29       0| Creating H264 Source...                                           
[main thread     ]ServerMediaSubsession.c:51       0| Creating Sink for format : video/H264                             
[main thread     ]ServerMediaSubsession.c:25       0| Creating Source...                                                
[main thread     ]ServerMediaSubsession.c:29       0| Creating H264 Source...                                           
[main thread     ]ServerMediaSubsession.c:51       0| Creating Sink for format : video/H264                             
[main thread     ]       DeviceSource.cpp:110           0| V4L2DeviceSource::doStopGettingFrames

esarigul, 2019-05-01:
I cannot set up wifi network either it stucks in setup mode when I open its webpage it show yellow notification message to set the camera everytime I connect to its web page ( video shows with problem), when I connect to openfang wifi network

anmaped, 2019-05-05:
@esarigul You have to configure the camera to your setting properly. Without telling us winch model are you using will be difficult since you have to write the correct nvram settings in order to the setup mode be concluded. Otherwise, you will not conclude the setup and then, no settings ca be stores in nvram. This is a security procedure to avoid overlapping an incorrect nvram position.

esarigul, 2019-05-05:
@anmaped thank you very much for your response, I am still working on it, everything in my camera seems to be different, flash partition layout, partition names and even flash size (8M instead of 16M), codes for nvram.c and its associated headers needs to be changed ( I am doing it those) as well, once I get everything straight I will let you know (btw in your branch your init scripts refers some scripts that are not exist in the branch e.g. rtsp-h264 and rtsp-mjpeg), and rtsp v0.0.8 works VLC player I have in my phone has some problems, tinyCAM or other rtsp player works very well. I can consider the issue is closed, but if I have I wil ask :-), take care

closed 2019-05-05 by esarigul

anmaped, 2019-05-06:
Are you designing something from scratch?

esarigul, 2019-05-06:
Andre you could say that my camera is haosihd which is not well known brand
like wyze, It just uses T20 SoC, I am trying to adapt and make your github
code work with my camera, I hate cloud services which handles my data, that
is main reason I am working on this project, it alse teach me embedded
linux which is a big plus for me :-)

anmaped, 2019-05-06:
Interesting, so use the gitter channel for any issue.