anmaped/openfang #34: [T20L Xiaomi Mijia 2018] Settings are gone after every reboot

martin-schlossarek, 2019-05-09:
No matter how I set the settings, whether via the web interface or via `nvram set`, they disappear after every reboot. Is this an expected behavior?

anmaped, 2019-05-09:
No. What camera are you using and version ?

martin-schlossarek, 2019-05-09:
I'm using Mijia 2018.

```
$ cat /opt/version 
rc05_01-46-g5dd319b
```

anmaped, 2019-05-09:
You are using a old version, I recommend you to compile the last one or wait for rc6. I'll provide it soon.
But as far as I know this version should be working well, try to setup the model, reboot, and setup the model again and reboot, it should solve your issue.

martin-schlossarek, 2019-05-10:
Hmm.. rc05_01-46-g5dd319b is from third most recent commit (5dd319b8390b8961f4ccdab9ff43c43d84c4ed36).

I built a new image from the latest commit (b1c396e2c087abf78fef47c629da56b9a551a613), but this version doesn't boot. The LED flashes orange (sometimes also blue), stays orange for 3 seconds and then goes out. The whole thing then repeats itself again and again.

anmaped, 2019-05-11:
I will introduce a fix for old bootloaders, but this is due to the lack of the last bootloader that includes the auto device detection.

anmaped, 2019-05-11:
Or, you can also add to the `uEnv.txt` file the variable `hwversion=SXJ02ZM`.

martin-schlossarek, 2019-05-13:
Still no success.

5dd319b with hwversion=SXJ02ZM bootargs parameter: settings gone after reboot
b1c396e with hwversion=SXJ02ZM bootargs parameter: not bootable

anmaped, 2019-05-13:
b1c396e you have to set up `hwversion=SXJ02ZM` again, since the uEnv.txt file is replaced by a fresh one.

anmaped, 2019-05-15:
Have it solved your problem?

martin-schlossarek, 2019-05-15:
I will try it again this evening because I'm not home right now. I'll keep you posted.

martin-schlossarek, 2019-05-15:
Ok, I added `hwversion=SXJ02ZM` to bootargs twice and now it's persisent. But still no look with `nvram set`. Everything is lost after a reboot:

[image]

anmaped, 2019-05-15:
You have to use 'nvram commit' to turn it persistent.

martin-schlossarek, 2019-05-15:
Yes... forgot to use `nvram commit` for the screenshot, but I already tried it with no luck:

[image]

Changes on the web ui also don't survive a reboot.

anmaped, 2019-05-16:
Try 'nvram clear rtdev; nvram commit; reboot' to factory reset the settings. The certificate should also be renewed by this action.

HorseDickJoseph, 2019-05-16:
Hi anmaped!

I have same issue. Is it poosible make revert back to stock?

anmaped, 2019-05-16:
@HorseDickJoseph Can you describe which issue you have?

anmaped, 2019-05-16:
You can also try to clean, the mtd partition using
`flash_eraseall /dev/mtdX`
where `X` is the number of the config mtd partition that you can check using `cat /proc/mtd`.

I tried to replicate the issue and I was not able to get it in the same camera and different models.

martin-schlossarek, 2019-05-16:
I tried both approaches `nvram clear rtdev; nvram commit` and `flash_eraseall /dev/mtd4`. Still no success. 

I followed the installation instructions from TheRoss and desoldered a leg of the SOP8 chip so I could flash the bootloader with a programmer clip. Then I soldered the leg back on. I booted the camera, opened the web ui, changed the wifi settings and the admin password and restarted the camera. When the camera didn't connect to the wifi network, started an AP again and the admin password didn't change, I noticed that no settings are saved at all.

Is it possible that the chip was somehow damaged during the soldering process so that it only runs in "read only" mode?

anmaped, 2019-05-16:
@martin-schlossarek It's strange, I'll upload the image and the bootloader and you can flash them and try (you can also try the gitter chat). Here I can't reproduce such issue, but let's see what @HorseDickJoseph has to say (he said same issue).

HorseDickJoseph, 2019-05-17:
@anmaped I have a sami issue, like @martin-schlossarek. After I saved the settings and reboot, lost everything what  Im set it. Looks like after the reboot, this is the first reboot after the flashing.

HorseDickJoseph, 2019-05-27:
@anmaped You have any news? How can I revert to stock the camera firmware? (I read the FAQ, if I remove the SD card, normal boot starting. But the camera isnt start.)
You have the original bootloader? My mistake, I forgot saving first :(

HorseDickJoseph, 2019-05-28:
Today arrived my second camera (same). I save the bootloader, and write back to the old one. Write is successful, but the camera wont boot.....
[original.zip](https://github.com/anmaped/openfang/files/3226982/original.zip)

anmaped, 2019-06-03:
you have the full dump on https://github.com/anmaped/openfang/blob/master/doc/sxj02ZM/dump.tar.xz

anmaped, 2019-06-03:
@martin-schlossarek I dind't understood your problem yet.

@martin-schlossarek Any news ? Have you solved the problem ?

martin-schlossarek, 2019-06-03:
> @martin-schlossarek Any news ? Have you solved the problem ?

No sorry I was not able to solve the problem. Is there anything I can do to find out what's going wrong? Logs, debug data ...?

HorseDickJoseph, 2019-06-05:
> @HorseDickJoseph I didn't understood your problem.
> 
> You have the full dump in https://github.com/anmaped/openfang/blob/master/doc/sxj02ZM/dump.tar.xz since December 2018. You can use it.
> 
> Edit: you can also find the unpacked FW in https://github.com/anmaped/openfang/tree/master/doc/sxj02ZM/unpacked

Can you tell me how can I use this files?

QuantumSingularity, 2019-12-26:
I unfortunely have to same problem.
The only setting that remains is het ntp-server..
Setting via the nvram set command and nvram commit doesn't persist either.
/opt/version == 2018-11-21 02:01 (latest version on https://github.com/anmaped/openfang/releases )

It's a v3 version of the camera and I've flashed the bootloader with the programmer and de-soldered the vcc-pin (and re-soldered it again after flashing).

QuantumSingularity, 2019-12-28:
I have found out a work-a-round.

I changed the settings in the webinterface.
After that I check the nvram with:
nvram layout
Then per section:
nvram show uboot
nvram show 2860
nvram show rtdev
nvram show cert
nvram show wapi

After that I edited the init.d file /etc/init.d/S02factory on the sdcard.

```
$ cat S02factory 
#!/bin/sh

case "$1" in 
	start)
		section_names=$(nvram layout | sed -n '/name:/p' | awk -F' ' '{print $3}')
		factory=$(nvram get rtdev factory >&2)
		if [ $? != 0 ]; then

			for sec in $section_names
			do
				echo "Clearing sec: $sec"
				nvram clear $sec
			done

			#nvram set rtdev factory 1
			#nvram set rtdev fw_version "$(cat /opt/version)"
			#nvram set wapi adminpasswd "admin"

			# Create a SSL certificate
			#export OPENSSL_CONF=/etc/ssl/openssl.cnf
			#openssl req -new -x509 -keyout /etc/ssl/lighttpd.pem -out /etc/ssl/lighttpd.pem -days 365 -nodes -subj "/C=DE/ST=.../L=.../O=.../OU=.../CN=.../emailAddress=..."
			#chmod 400 /etc/ssl/lighttpd.pem

			#nvram set rtdev certificate "$(cat /etc/ssl/lighttpd.pem)"

			nvram set rtdev factory 0
			nvram set rtdev fw_version "$(cat /opt/version)"
			# restore the SSL certifcate
			nvram set rtdev certificate "$(cat /etc/ssl/lighttpd.pem)"
			nvram set rtdev model "T20L_M"
			nvram set rtdev motion_script_on "sh /opt/scripts/motion-script-on.sh"
			nvram set rtdev motion_script_off "sh /opt/scripts/motion-script-off.sh"
			nvram set rtdev motor false

			nvram set 2860 blue_led_pin "led_blue1"
			nvram set 2860 yellow_led_pin "led_yellow1"
			nvram set 2860 ir_pin 72

			nvram set wapi adminpasswd "<<your password>>"
			nvram set wapi mac "<<your_mac>>"
			nvram set wapi wmode "STA"
			nvram set wapi wssid "<<your ssis>>"
			nvram set wapi wpassword "<<your password>>"
			nvram set wapi wkeymgmt "KEY"

			nvram commit

		else
			[ "$(nvram get rtdev fw_version)" != "$(cat /opt/version)" ] && nvram set rtdev fw_version "$(cat /opt/version)"
			
			# restore the SSL certifcate
			nvram get rtdev certificate > "/etc/ssl/lighttpd.pem"
			chmod 400 /etc/ssl/lighttpd.pem
		fi
		;;
	stop)
		echo "do nothing"
		;;
	restart|reload)
		"$0" stop
		"$0" start
		;;
	*)
		echo "Usage: $0 {start|stop|restart}"
		exit 1
esac

exit $?
$ 

```