anmaped/openfang #35: default nvram settings of SXJ02ZM

mgx0, 2019-05-14:
Hi,

does anybody have a possibility to export nvram settings (not the certificate, that can be recreated)?

I flushed nvram and lost the pin assignments. Since then also wifi stopped working (connects and gets IP address, but does not communicate)

thanks

anmaped, 2019-05-15:
'nvram clear rtdev; nvram commit' and do a reboot! This will factory reset the device!

mgx0, 2019-05-16:
thanks, did that, I see the AP now (it was not visible not the wifi did not respond to anything), but I have lost the pin mapping for example to blue and orange led, IR and cutoff filter pins are not defined anywhere. or are these nvram entries not needed for openfang?

closed 2019-05-16 by mgx0