anmaped/openfang #39: [T20L Xiaomi Mijia 2018] Auto Night detection

lenisko, 2019-07-02:
Looks like it's not working
```
$ autonight -D /dev/jz_adc_aux_0 -v -d 1
Current value: 710.48
Current value: 710.12
Current value: 710.24
Current value: 710.72
Current value: 714.32
Window (5) Avg: 711.18
Current value: 711.20
Window (5) Avg: 711.32
Current value: 711.68
Window (5) Avg: 711.63
Current value: 710.00
Window (5) Avg: 711.58
Current value: 714.80
Window (5) Avg: 712.40
```

I have cowered cam with my hand in half of those readings.
Values are very odd...

Any clues?

eteixido1, 2019-08-14:
Same values for me with covered and uncovered camera (between 719 and 717). Currently I'm using my own compiled version and night mode doesn't activate.

eteixido1, 2019-08-14:
Hello Lenisko, today afer starting IR LED and IR CUT, autonight start returning values between 1800 & 50000. Maybe autonight needs one of them.

eteixido1, 2019-09-10:
I don't know if there are many options in autonight mode. Some times doesn't start and some times IR-CUT remains active. In general IR-CUT + IR-LED looks better than night mode.

minhan74, 2020-01-12:
My auto night mode doesn't work too. I'm using the Mijia v3 2018 with the build 6117b8d

I tried to see the value on the terminal and this is result:

$ autonight -D /dev/jz_adc_aux_0 -v -d 1
Current value: 0.00
Current value: 0.00
Current value: 0.00
Current value: 0.00
Current value: 0.00
Window (5) Avg: 0.00
Night Mode Enabled
failed to load /tmp/.nvramcache, Permission denied, nvram_parse
/opt/scripts/nightmode.sh: line 5: can't create /sys/class/gpio/gpio1/value: nonexistent directory
failed to load /tmp/.nvramcache, Permission denied, nvram_parse
/opt/scripts/nightmode.sh: line 5: can't create /sys/class/gpio/gpio1/value: nonexistent directory
failed to load /tmp/.nvramcache, Permission denied, nvram_parse
/opt/scripts/nightmode.sh: line 5: can't create /sys/class/gpio/gpio0/value: nonexistent directory
failed to load /tmp/.nvramcache, Permission denied, nvram_parse
/opt/scripts/nightmode.sh: line 5: can't create /sys/class/gpio/gpio0/value: nonexistent directory
/opt/scripts/nightmode.sh: line 5: can't create /var/run/ircut: Permission denied
Bus error
WARNING: /opt/scripts/nightmode.sh on
 returned 35328

Please help me on how to grant permission? Thank you.

goncalossilva, 2020-08-16:
Also not working on 695670d5977bcb7b77c4fcbadeb9885c08546743 (latest master).

axlerose, 2020-09-25:
Any news about this issue?
Is the project alive?

americodias, 2020-10-26:
> My auto night mode doesn't work too. I'm using the Mijia v3 2018 with the build [6117b8d](https://github.com/anmaped/openfang/commit/6117b8d9873ac0ab7f67ccc7c27fca077d6a08e4)
>
> I tried to see the [...]

Hi @minhan74!

Did you fixed this? I have the same problem.

Thanks,
Américo

axlerose, 2020-10-26:
Hi, I've the same problem.have anyone fix it?

fastiuk, 2022-06-09:
There is no fix. Xiaomi Mijia 2018 v3 doesn't have a light sensor. I opened it and checked.
The only solution here is to use "autonight -S" in "/opt/controlscripts/auto-night-detection" script.