anmaped/openfang #40: [T20L Xiaomi Mijia 2018] Revert back to stock

irakic89, 2019-07-05:
I tried reverting back to stock firmware by removing SD card as suggested in FAQ, but nothing happens, LED doesn't light up at all.

How can I flash it back to stock firmware? Recovery file on SD root didn't help.

Thanks

anmaped, 2019-09-10:
@irakic89 Have you solved the issue ?

flexxo, 2019-11-17:
I think the best way to revert back to stock is to open camera again and flash the original image into the 25 spi flash. But nobody seems to share their original dump, If original dumps are not unique to each device, I think it maybe a todo for this project to include some original firmware for those who want to revert.
Yes, unfortunately I didn't make backups of my own too :(

amricko0b, 2019-11-23:
Greetings!

Seems like the only way of reverting (according to @flexxo) must be mentioned in installation guide.

irakic89, 2020-02-18:
Does anyone have a stock backup to share? It seems to me that in the current state camera ispretty much unusable. 
Thanks

ciB89, 2020-04-28:
Also interested in the original firmware.. The camera just keeps crashing / the stream stops after some time and settings are not properly saved..