anmaped/openfang #47: Need help flashing SXJ02ZM

ciB89, 2019-10-28:
So I followed the guide specific for the SXJ02ZM, but I am unsure at a few points:
In the guide it says flashing with windows does not work, but then it gives instructions on how to flash on windows. I followed the instructions and flashing _seemed_ to work (the chip is not empty any more afterwards). So, did that work?

Secondly, do I resolder the desoldered leg after I am finish flashing? Do I need to reasemble the whole camera, or should it work with only the mainpiece for testing?

I would not be asking all these questions if my cam worked, so I'm hoping somebody can clarify. Currently I powered up the cam after flashing with windows and resoldering the leg, but the cam does not seem to draw power since no led flashes.

mgx0, 2019-10-28:
not sure about the windows flashing part.

but yes you have to solder it back where it was soldered. You don't need to reassemble it back. just make sure it is connected just like it was before (for example you must solder the leg back)

ciB89, 2019-10-28:
Apparently I did not solder the leg correctly, now its turning on. The LED does not start to flash though, it's just shining in a whitish-purple collor. And there's no WIFI hotspot

ciB89, 2019-10-28:
Reflashed the SD-Card and now it works :)

closed 2019-10-28 by ciB89

sugar012, 2023-10-16:
Hello there. In 2023 i'm trying to revive this cams since i have alrady recording in the house but i want to integrate them into frigate. I tried flashing with windows and all seemed succesfull but when i power it up nothing happens. i used the older release RC4 that apparenly is for 64mb version? The filename is the same as RC5 i tested pin8 and it draw 3.3v once powered up so maybe is not a re-soldering problem. 

Any idea how to debug?