anmaped/openfang #56: Boot loop on Wyze cam 2

xdissent, 2020-03-10:
I'm running openfang master (695670d) on a Wyze Cam 2 and [this check](https://github.com/anmaped/openfang/blob/695670d5977bcb7b77c4fcbadeb9885c08546743/fs/etc/init.d/S02factory#L36) is sending it into a boot loop. On my camera, the reset button gpio (pin 50) is always `0` at this point in the boot process, regardless of whether I'm pressing it or not. I don't know if it's looking at the wrong pin or what, but my camera always clears `rtdev` nvram and reboots, flip flopping between factory reset mode and this check. Disabling the check lets me boot fine.

xdissent, 2020-03-10:
I think this is the behavior #51 describes

anmaped, 2020-03-10:
@xdissent Did you flash the compiled openfang bootloader?

xdissent, 2020-03-11:
Yes, a compiled openfang bootloader (from master, built in docker locally) has been flashed.

tachang, 2020-07-12:
@xdissent When you compiled and flashed the bootloader how large was the file? Is there a reason you didn't use the released one?