anmaped/openfang #57: SD card init fails every other boot

xdissent, 2020-03-10:
I don't really follow the "ping/pong" sd card initialization seen [here](https://github.com/anmaped/openfang/blob/695670d5977bcb7b77c4fcbadeb9885c08546743/patches/uboot/uboot-isvp_t20_board.c.patch#L28) and [here](https://github.com/anmaped/openfang/blob/695670d5977bcb7b77c4fcbadeb9885c08546743/patches/uboot/uboot-isvp_t20.h.patch#L54) but it's failing to enable SD ever other boot. 

A successful boot looks like this for me:
```
--->probe spend 5 ms
Enables SDCARD
ping
gpio_request lable = cmd_gpio gpio = 43
gpio: pin 43 (gpio 43) value is 1
gpio_request lable = cmd_gpio gpio = 48
gpio: pin 48 (gpio 48) value is 0
gpio_request lable = cmd_gpio gpio = 43
gpio: pin 43 (gpio 43) value is 0
Net:   Jz4775-9161
Hit any key to stop autoboot:  1  0 
soctype=T20X
Settings are already stored!
Valid chip addresses:
count=1
ping: 1
the manufacturer 1c
SF: Detected EN25QH128A

Erasing SPI flash...Writing to SPI flash...done
gpio_request lable = cmd_gpio gpio = 43
gpio: pin 43 (gpio 43) value is 1
gpio_request lable = cmd_gpio gpio = 48
gpio: pin 48 (gpio 48) value is 0
gpio_request lable = cmd_gpio gpio = 43
gpio: pin 43 (gpio 43) value is 0
Hardware found: WYZEC2
MMC Found
440 bytes read in 33 ms (12.7 KiB/s)
1781202 bytes read in 150 ms (11.3 MiB/s)
## Booting kernel from Legacy Image at 80600000 ...
   Image Name:   Linux-3.10.14
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1781138 Bytes = 1.7 MiB
   Load Address: 80010000
   Entry Point:  803c9e30
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
```
followed by:
```
[    1.459079] Waiting for root device /dev/mmcblk0p1...
[    1.794322] dwc2 dwc2:  ++OTG Interrupt: A-Device Timeout Change++
[    1.949182] mmc0: new SDHC card at address 0001
[    1.954202] mmcblk0: mmc0:0001 00000 14.5 GiB 
[    1.960143]  mmcblk0: p1 p2
[    2.014158] EXT4-fs (mmcblk0p1): mounting ext3 file system using the ext4 subsystem
[    2.297017] EXT4-fs (mmcblk0p1): recovery complete
[    2.306304] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
```

The next boot always looks like this:
```
--->probe spend 5 ms
Enables SDCARD
pong
gpio_request lable = cmd_gpio gpio = 43
gpio: pin 43 (gpio 43) value is 1
gpio_request lable = cmd_gpio gpio = 48
gpio: pin 48 (gpio 48) value is 0
Net:   Jz4775-9161
Hit any key to stop autoboot:  1  0 
soctype=T20X
Settings are already stored!
Valid chip addresses:
count=2
pong: 2
the manufacturer 1c
SF: Detected EN25QH128A

Erasing SPI flash...Writing to SPI flash...done
gpio_request lable = cmd_gpio gpio = 43
gpio: pin 43 (gpio 43) value is 1
gpio_request lable = cmd_gpio gpio = 48
gpio: pin 48 (gpio 48) value is 0
Hardware found: WYZEC2
MMC Found
440 bytes read in 35 ms (11.7 KiB/s)
1781202 bytes read in 153 ms (11.1 MiB/s)
## Booting kernel from Legacy Image at 80600000 ...
   Image Name:   Linux-3.10.14
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1781138 Bytes = 1.7 MiB
   Load Address: 80010000
   Entry Point:  803c9e30
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
```
and then it hangs at:
```
[    1.459076] Waiting for root device /dev/mmcblk0p1...
[    1.794332] dwc2 dwc2:  ++OTG Interrupt: A-Device Timeout Change++
```

The LED never shows any blue on a failed boot, just flashing yellow.

From what I can tell, `count: 1` means it works, `count: 2` means it hangs. [Here](https://github.com/anmaped/openfang/blob/695670d5977bcb7b77c4fcbadeb9885c08546743/patches/uboot/uboot-isvp_t20.h.patch#L56) it looks like if `count :1` it immediately sets `count: 2`. Any idea why?

anmaped, 2020-05-20:
Could you check line 50?

`elseiftest\"${soctype}\"=\"T20X\"&&test\"${flashtype}\"=\"EN25QH128A\"&&i2cprobe0x40;then\+`

Try to remove this part `&& i2c probe 0x40`.

Problem: the bootloader didn't identify more than two times the same ' if then ' and then the ping pong starts.

Ozzyminted, 2020-05-21:
I got this a lot with rc5 release. Never managed to solve it. And never had
any luck on my neos building the master branch.

Be good if some others got involved to get the project moving again. Its a
shame its stagnated.