anmaped/openfang #58: Boot loop on SXJ02ZM

mje-nz, 2020-03-22:
I'm running master ([9d34772](https://github.com/anmaped/openfang/commit/9d3477229108f4bb1f7694fa5b01e03faf8305ec)) on a Xiaomi Mijia 2018, and it's stuck in a boot loop. It's not the same issue as #56 -- it always gets past the factory reset check and doesn't restart until all the init scripts have run. It doesn't sound like #51 either -- I do get the LED blinking orange and blue and the AP showing up, but it boot loops without stopping and it's never wiped the SD card.

Here are the serial logs for the first three boots from a fresh install:
[master-boot.log](https://github.com/anmaped/openfang/files/4364483/master-boot.log)

@anmaped Do you have a known-working commit for SXJ02ZM I can go back to?

llimz, 2020-05-08:
I've the exact same issue here with the same IP Cam model. Have you found a way to solve it? Thank you

llimz, 2020-05-12:
For information, I built a new bootloader from scratch by using the last source code on GitHub. I flashed again my ipcam and the problem is now solved. It boots correctly, and I can reach it throught the openfang wireles AP.

So for me the last bootloader solves the problem.

CamFlyerCH, 2020-05-19:
> So for me the last bootloader solves the problem.

I have the tools to program the flash and the cam open, but I am not able to compile this. I admit, I am a Windows guy. I have a hard time when I need to work with linux. So I would be really happy if someone can put the binaries for the bootloader and the SD card somewhere or even create an release that should work with this cam. Thanks !

llimz, 2020-05-20:
This is link where you can download the bootloader I built (8th of May 2020)

https://www.icloud.com/iclouddrive/0H8if7EBB_qxo03mdVsmtu-hQ#u-boot-lzo-with-spl%5Ft20%5F64M

anmaped, 2020-05-21:
@llimz Many thanks. I'm finishing the CI for doing the releases automatically. I didn't had opportunity to finish the auto update but I'll enable it asap since many people is using it for SXJ02ZM.

I would like to ask for testers and do a small list.

CamFlyerCH, 2020-05-21:
> This is link where you can download the bootloader I built (8th of May 2020)
> 
> https://www.icloud.com/iclouddrive/0H8if7EBB_qxo03mdVsmtu-hQ#u-boot-lzo-with-spl%5Ft20%5F64M

Thank you very much ! I would like the SD-Card Image to if possible.

llimz, 2020-05-21:
Here you are. 

https://www.icloud.com/iclouddrive/0m5dK_4geRcueh9aKBY2ezBXQ#rootfs.ext2

@anmaped With pleasure, add me to your testers list !

ciB89, 2020-05-21:
> I would like to ask for testers and do a small list.

I'd be willing to test any future releases for the SXJ02ZM model.

Ozzyminted, 2020-05-21:
Think this will fix boot loop on neos/wyze also? Where did you compile the
bootloader and rootfs images from btw?

llimz, 2020-05-22:
I don't know if it will fix the boot loop on neos/wyze. I don't have those cameras.

I built the bootloader and rootfs by using the git reposiry. You can find more details here https://github.com/anmaped/openfang/blob/master/doc/docker.md.

Ozzyminted, 2020-05-22:
weird that's what I did. will give it another  go on fresh system.

jakass202, 2020-06-03:
@Ozzyminted did you manage to fix boot loop on neos?

djtomasnl, 2020-06-04:
Im new, but i succeeded to flash my xiaomi 2018 cam! Only thing difficulty is to compile the files, so thank you for putting the files in here!
It is not possible to update the openfang software at the webinterface. Do i need to flash the camera again? Or is it just enough to put the new software on the SD card?
Cam is working great, but changes its IP adress randomly (also mac adres?).. so, i hope that your files will fix this issue.
Thank you in advance!

CamFlyerCH, 2020-06-06:
> Here you are.
> 
> https://www.icloud.com/iclouddrive/0m5dK_4geRcueh9aKBY2ezBXQ#rootfs.ext2
> 
So I got the firmware running ! Thanks !

But I now find so many problems with it and I am wondering, of these are know problems or just in my setup. (Some of them forced me to reformat and redo the SD card. This changed also the IP !?!

- 1080 resolution not working
- microphone / audio
- storing LED settings
- Auto IR/Nightmode
- IR cut always on after reboot
- NTP time sync (it is always on UTP despite all my TZ strings I tested.)
- Terminal on web GUI (not entry possible, just static, Via SSL (Putty) it works)
- Timelapse images are 0 bytes in size
- Firmware update in GUI

What er the projects goals / timeline ?

Ozzyminted, 2020-06-21:
Tbh, i think i am going to reflash back to dafang-hacks. That repo is heavily developed and supported atm. Just need to put the old bootloader back

ciB89, 2020-06-21:
> Tbh, i think i am going to reflash back to dafang-hacks. That repo is heavily developed and supported atm. Just need to put the old bootloader back

Is it possible to run dafang-hacks on this camera? I was also thinking about it but was not sure how to accomplish this.

eu-smirnov, 2020-06-23:
> * 1080 resolution not working
> * microphone / audio
> * storing LED settings
> * Auto IR/Nightmode
> * IR cut always on after reboot
> * NTP time sync (it is always on UTP despite all my TZ strings I [...]

I can add to the list that I haven't succeeded in connecting to 5GHz WiFi, and that the time string has completely disappeared from the OSD (probably a reboot will fix this).

To be honest, I don't think we can expect a well-defined project timeline, as it's obvious that this is developed as a hobby project.
And many thanks to the author that this project even exists and works.

But I do think that a bit more information from @anmaped on the current priorities and issues, together with his thoughts on how they could be solved, would be really helpful to get and can make the project move forward faster.

For example, I'm going to look next (when I have time, of course) into the issue with TZ/time OSD, as it annoys me the most.