anmaped/openfang #65: Wyze Cam V3

pranas, 2020-11-02:
Wyze recently announced the updated [Wyze Cam V3](https://wyze.com/wyze-cam-v3.html). It looks like a great improvement: f/1.6 aperture, 2x processing power. Anyone got any more info on these? Are they still based on common hardware? What are other products with the same hardware then? What are the capabilities of flashing open source firmware on them?

idarlund, 2021-04-08:
It's a T31 chip
https://fccid.io/2AUIUWYZEC3/Internal-Photos/Internal-Photos-5001016

anmaped, 2021-04-09:
Could someone be interested in sending me a wyzev2 and wyzev3 (t31)
(europe) to enable openfang on these devices?

Thanks.

SomebodySysop, 2021-04-12:
> Could someone be interested in sending me a wyzev2 and wyzev3 (t31) (europe) to enable openfang on these devices? Thanks.
> […](#)

I can send you a V2.  How do I get it to you?

anmaped, 2021-04-12:
Use PM on openfang gitter channel.

AndrewBreyen, 2021-12-03:
Sorry to bump a possibly stale issue, but has there been updates on this? 
I'm extremely interested in using this for my v3 cams

SomebodySysop, 2021-12-03:
> Sorry to bump a possibly stale issue, but has there been updates on this? I'm extremely interested in using this for my v3 cams

Wyze has released, last I heard,  beta RTSP firmware for their V3 cameras: https://support.wyze.com/hc/en-us/articles/360026245231-Wyze-Cam-RTSP

There is also a github project which allows you to expose an RTSP stream from your V3 cameras without using customized firmware: https://github.com/mrlt8/docker-wyze-bridge